diyless/opentherm-thermostat3 #2: firmware code? or even pin-out?

aleks-mariusz, 2024-11-29:
I'm curious if the code-base that the thermostat3 runs is available (is it open?).. is it by any chance [this one](https://github.com/diyless/esp32-wifi-thermostat/blob/main/esp32-wifi-thermostat/esp32-wifi-thermostat.ino) ? 

I'm curious also what the pin-out is on the small 8-pin connector (2 rows of 4-pins).  

My boiler (intergas combi hre 36/30) also has a PC connection terminal, and per [this project](https://github.com/rixvet/IntergasBoilerReader/tree/trunk) it has information that is available via a serial interface.

would be great if i could connect this up to the opentherm-thermostat3 on one of those pin connectors and be able to view the data being sent via that interface

diyless, 2024-12-01:
Hi
We will update the product page to include pinout for that connector.
However, you won't be able to use it for communication though, since it is used for OpenTherm, power, and the external sensor. In theory boiler's TX can be connected instead of the Ext Sensor, however that pin is routed to the co-processor (rather than ESP32 itself), responsible for the OpenTherm and onewire sensor interfacing. It would not be able to read serial data from that (free) pin without significant software modification. Also, it is not clear if boiler's serial interface needs some galvanic isolation or not.

aleks-mariusz, 2024-12-02:
Awesome thank you for the explaination.  It makes perfect sense.

Ok so the easily-accessible connector is unlikely to be useful.  What about if one disassembles the unit, are there somewhat possible alternatives on the actual board inside one could 'tap' into without too much difficulty for the serial interface?

Also, as far as utilizing that project referenced that can decode the communication, what would be needed for that, code modification?  Guessing that would require the esp32 code-base being opened up?

diyless, 2024-12-02:
ESP32 has no free IO left at all. They have been eaten by the RGB display, even OpenTherm is dispatched to a separate SoC. The only possibility is to modify **both** ESP32 and STM32 firmware, so STM32 pulls the data from the boiler, and ESP32 understands that STM32 got a new feature and is able to deal with it. However that would require PCB change, since for now only a needed IOs of the STM32 are routed somewhere.
Alternatively, you can fire up just a plain ESP32 controller with i.e. esphome. It is pretty configurable to allow you achieve your goal.

aleks-mariusz, 2024-12-02:
Gotcha, ok that makes sense.  I thought it would be a good chance to use the opentherm-thermostat3 device for this, but it sounds like it's already setup in a way that uses up all its io connectivity and is highly unfeasible to have this added functionality added to it.  I'll look into a separate esp32 w/ esphome to look into this at some point.  Thanks for the details so far! 

I'll close this issue as i'm out of questions for once :-_) and whenever you get the pinout added, great..

closed 2024-12-02 by aleks-mariusz

diyless, 2024-12-02:
We can add some free-to-use pins for the STM32, however writing a firmware for that wouldn't be an easy walk.