diyless/opentherm-thermostat3 #3: HA log entries: Can't decode payload

aleks-mariusz, 2024-12-16:
I am seeing this in homeassistant regularly:

```
Can't decode payload b'{"idx":"diyless_E4B06381B4C4","temperature":"21.1216","temperatureRaw":"21.1307","targetTemperature":24,"mode":"heat","targetTemperature_manual":10,"mode_manual":"off","action":"idle","action_manual":"idle","CHEnabled":true,"CHState":true,"CHTemperature":70.31641,"CHPressure":2.058594,"CHRetTemp":0,"DHWState":false,"temperature_manual":"\x88G\xd5\xb0","DHWAction":"idle","DHWTemperature":48.10938,"DHW2Temperature":0,"OutsideTemp":0,"ExhaustTemp":0,"FlameState":false,"FlameLevel":0,"FaultState":false,"FaultCode":0,"DiagnosticState":false,"CHSetpoint":80,"chMin":0,"chMax":0,"chBoilerUpper":80,"chBoilerMax":80,"DHWMode":"heat","DHWEnabled":true,"DHWSetpoint":60,"IntegralTerm":48.02654,"IntegralTermRestored":true,"SensorReady":true,"SensorError":false,"OTConnLoss":false,"OTConnState":true,"SensorInt":21.1216,"bleBatt":null,"bleBattLevel":null,"bleHumidity":null,"bleTemp":null,"blePacketLossRatio":null,"bleRssi":null,"bleLastSeen":null,"mqttSenState":false,"mqttSenConnState":null,"mqttSenTemp":null,"mqttSenLastSeen":null,"intSenState":true,"intSenConnState":true,"intSenTemp":21.1216,"WiFiRssi":-45}' on diyless_thermostat/stat/things/thermostat/properties with encoding utf-8 (for <Job None HassJobType.Callback functools.partial(<bound method MqttEntity._message_callback of <entity binary_sensor.intergas_boiler_mqtt_sensor_connection_state=unavailable>>, <bound method MqttBinarySensor._state_message_received of <entity binary_sensor.intergas_boiler_mqtt_sensor_connection_state=unavailable>>, {'_expired', '_attr_is_on'})>)
```

it says: `First occurred: 2:16:12 PM (105 occurrences)` (i cleared logs around this time to see if it was a one off or it would come back).

to me It looks like the temperature_manual value has unicode characters which may not be being handled correctly?

diyless, 2024-12-17:
> "temperature_manual":"\x88G\xd5\xb0"

that looks definitely strange. will take a look on that value initialization

aleks-mariusz, 2025-01-02:
FWIW, i also see this occurring with the `temperatureRaw` field:

```
Can't decode payload b'{"idx":"diyless_E4B06381B4C4","temperature":"20.4081","temperatureRaw":",\xf3\xac!\x96\x01","targetTemperature":22.5,"mode":"heat","targetTemperature_manual":10,"mode_manual":"off","action":"idle","action_manual":"idle","CHEnabled":true,"CHState":true,"CHTemperature":54.53906,"CHPressure":2.109375,"CHRetTemp":0,"DHWState":false,"temperature_manual":"54.5","DHWAction":"idle","DHWTemperature":46.76953,"DHW2Temperature":0,"OutsideTemp":0,"ExhaustTemp":0,"FlameState":false,"FlameLevel":0,"FaultState":false,"FaultCode":0,"DiagnosticState":false,"CHSetpoint":80,"chMin":0,"chMax":0,"chBoilerUpper":80,"chBoilerMax":80,"DHWMode":"heat","DHWEnabled":true,"DHWSetpoint":60,"IntegralTerm":58.18122,"IntegralTermRestored":true,"SensorReady":true,"SensorError":false,"OTConnLoss":false,"OTConnState":true,"SensorInt":20.40806,"bleBatt":null,"bleBattLevel":null,"bleHumidity":null,"bleTemp":null,"blePacketLossRatio":null,"bleRssi":null,"bleLastSeen":null,"mqttSenState":false,"mqttSenConnState":null,"mqttSenTemp":null,"mqttSenLastSeen":null,"intSenState":true,"intSenConnState":true,"intSenTemp":20.40806,"WiFiRssi":-44}' on diyless_thermostat/stat/things/thermostat/properties with encoding utf-8 (for <Job None HassJobType.Callback functools.partial(<bound method MqttEntity._message_callback of <entity binary_sensor.intergas_boiler_internal_sensor_state=on>>, <bound method MqttBinarySensor._state_message_received of <entity binary_sensor.intergas_boiler_internal_sensor_state=on>>, {'_attr_is_on', '_expired'})>)
```

diyless, 2025-01-02:
> ,\xf3\xac!\x96\x01

is definitely wrong value, will try investigating where does it came from

aleks-mariusz, 2025-01-09:
here some other types of errors, all seem like they're related to invalid values.. any ideas where they are originating from, or how to diagnose this on my side?

```
Invalid modes mode: {"idx":"diyless_E4B06381B4C4","temperature":"6ɡ","temperatureRaw":"20.6543","targetTemperature":22.5,"mode":"heat","targetTemperature_manual":10,"mode_manual":"off","action":"heating","action_manual":"heating","CHEnabled":true,"CHState":true,"CHTemperature":55.96875,"CHPressure":2.066406,"CHRetTemp":0,"DHWState":false,"temperature_manual":"56.0","DHWAction":"idle","DHWTemperature":47.29688,"DHW2Temperature":0,"OutsideTemp":0,"ExhaustTemp":0,"FlameState":true,"FlameLevel":7,"FaultState":false,"FaultCode":0,"DiagnosticState":false,"CHSetpoint":80,"chMin":0,"chMax":0,"chBoilerUpper":80,"chBoilerMax":80,"DHWMode":"heat","DHWEnabled":true,"DHWSetpoint":60,"IntegralTerm":68.75444,"IntegralTermRestored":true,"SensorReady":true,"SensorError":false,"OTConnLoss":false,"OTConnState":true,"SensorInt":20.63789,"bleBatt":null,"bleBattLevel":null,"bleHumidity":null,"bleTemp":null,"blePacketLossRatio":null,"bleRssi":null,"bleLastSeen":null,"mqttSenState":false,"mqttSenConnState":null,"mqttSenTemp":null,"mqttSenLastSeen":null,"intSenState":true,"intSenConnState":true,"intSenTemp":20.63789,"WiFiRssi":-43}
Invalid modes mode: {"idx":"diyless_E4B06381B4C4","temperature":"=%Ħ\u0000","temperatureRaw":"20.5739","targetTemperature":22.5,"mode":"heat","targetTemperature_manual":10,"mode_manual":"off","action":"heating","action_manual":"heating","CHEnabled":true,"CHState":true,"CHTemperature":74.80859,"CHPressure":2.179688,"CHRetTemp":0,"DHWState":false,"temperature_manual":"74.8","DHWAction":"idle","DHWTemperature":46.40625,"DHW2Temperature":0,"OutsideTemp":0,"ExhaustTemp":0,"FlameState":true,"FlameLevel":0,"FaultState":false,"FaultCode":0,"DiagnosticState":false,"CHSetpoint":80,"chMin":0,"chMax":0,"chBoilerUpper":80,"chBoilerMax":80,"DHWMode":"heat","DHWEnabled":true,"DHWSetpoint":60,"IntegralTerm":68.75444,"IntegralTermRestored":true,"SensorReady":true,"SensorError":false,"OTConnLoss":false,"OTConnState":true,"SensorInt":20.57514,"bleBatt":null,"bleBattLevel":null,"bleHumidity":null,"bleTemp":null,"blePacketLossRatio":null,"bleRssi":null,"bleLastSeen":null,"mqttSenState":false,"mqttSenConnState":null,"mqttSenTemp":null,"mqttSenLastSeen":null,"intSenState":true,"intSenConnState":true,"intSenTemp":20.57514,"WiFiRssi":-43}
Invalid modes mode: {"idx":"diyless_E4B06381B4C4","temperature":"TKuQ","temperatureRaw":"20.8602","targetTemperature":22.5,"mode":"heat","targetTemperature_manual":10,"mode_manual":"off","action":"idle","action_manual":"idle","CHEnabled":true,"CHState":true,"CHTemperature":68.39844,"CHPressure":2.117188,"CHRetTemp":0,"DHWState":false,"temperature_manual":"68.4","DHWAction":"idle","DHWTemperature":50.79688,"DHW2Temperature":0,"OutsideTemp":0,"ExhaustTemp":0,"FlameState":false,"FlameLevel":0,"FaultState":false,"FaultCode":0,"DiagnosticState":false,"CHSetpoint":80,"chMin":0,"chMax":0,"chBoilerUpper":80,"chBoilerMax":80,"DHWMode":"heat","DHWEnabled":true,"DHWSetpoint":60,"IntegralTerm":68.75444,"IntegralTermRestored":true,"SensorReady":true,"SensorError":false,"OTConnLoss":false,"OTConnState":true,"SensorInt":20.85209,"bleBatt":null,"bleBattLevel":null,"bleHumidity":null,"bleTemp":null,"blePacketLossRatio":null,"bleRssi":null,"bleLastSeen":null,"mqttSenState":false,"mqttSenConnState":null,"mqttSenTemp":null,"mqttSenLastSeen":null,"intSenState":true,"intSenConnState":true,"intSenTemp":20.85209,"WiFiRssi":-43}
Invalid modes mode: {"idx":"diyless_E4B06381B4C4","temperature":"19.8578","temperatureRaw":"19.8500","targetTemperature":22.5,"mode":"heat","targetTemperature_manual":10,"mode_manual":"off","action":"idle","action_manual":"idle","CHEnabled":true,"CHState":true,"CHTemperature":56.03906,"CHPressure":2.089844,"CHRetTemp":0,"DHWState":false,"temperature_manual":"tಹ","DHWAction":"idle","DHWTemperature":46.10938,"DHW2Temperature":0,"OutsideTemp":0,"ExhaustTemp":0,"FlameState":false,"FlameLevel":0,"FaultState":false,"FaultCode":0,"DiagnosticState":false,"CHSetpoint":80,"chMin":0,"chMax":0,"chBoilerUpper":80,"chBoilerMax":80,"DHWMode":"heat","DHWEnabled":true,"DHWSetpoint":60,"IntegralTerm":68.75444,"IntegralTermRestored":true,"SensorReady":true,"SensorError":false,"OTConnLoss":false,"OTConnState":true,"SensorInt":19.85779,"bleBatt":null,"bleBattLevel":null,"bleHumidity":null,"bleTemp":null,"blePacketLossRatio":null,"bleRssi":null,"bleLastSeen":null,"mqttSenState":false,"mqttSenConnState":null,"mqttSenTemp":null,"mqttSenLastSeen":null,"intSenState":true,"intSenConnState":true,"intSenTemp":19.85779,"WiFiRssi":-43}
Invalid modes mode: {"idx":"diyless_E4B06381B4C4","temperature":"19.7315","temperatureRaw":"19.7254","targetTemperature":22.5,"mode":"heat","targetTemperature_manual":10,"mode_manual":"off","action":"heating","action_manual":"heating","CHEnabled":true,"CHState":true,"CHTemperature":51.36719,"CHPressure":2.066406,"CHRetTemp":0,"DHWState":false,"temperature_manual":"ԁ0","DHWAction":"idle","DHWTemperature":43.58984,"DHW2Temperature":0,"OutsideTemp":0,"ExhaustTemp":0,"FlameState":true,"FlameLevel":55,"FaultState":false,"FaultCode":0,"DiagnosticState":false,"CHSetpoint":80,"chMin":0,"chMax":0,"chBoilerUpper":80,"chBoilerMax":80,"DHWMode":"heat","DHWEnabled":true,"DHWSetpoint":60,"IntegralTerm":68.75444,"IntegralTermRestored":true,"SensorReady":true,"SensorError":false,"OTConnLoss":false,"OTConnState":true,"SensorInt":19.73152,"bleBatt":null,"bleBattLevel":null,"bleHumidity":null,"bleTemp":null,"blePacketLossRatio":null,"bleRssi":null,"bleLastSeen":null,"mqttSenState":false,"mqttSenConnState":null,"mqttSenTemp":null,"mqttSenLastSeen":null,"intSenState":true,"intSenConnState":true,"intSenTemp":19.73152,"WiFiRssi":-43}
```

diyless, 2025-10-31:
@aleks-mariusz does the issue still reproduces? or can we close it?