diyless/opentherm-thermostat3 #4: Outdoor and boiler temperature is wrong

k-janssens, 2025-01-01:
Today I've replaced my Nest thermostat with the Thermostat3. 

Controlling CH and DHW seems to be working - will need to test further the coming days - but the outdoor and boiler temperature are -34C which is obviously wrong.
In the Nest thermostat, I could see the correct outdoor sensor value.

The boiler temperature did give some values but only briefly:
[image]

Does this have a negative impact on the working of my system?
And is there anything I can do to rectify this?

Some additional details:
[image]

aleks-mariusz, 2025-01-02:
> boiler temperature did give some values but only briefly

Perhaps i'm mistaken but this graph's title looks like it's for just the DHW (domestic hot water), so not the temperature of the CH (central heating).  It would only go up when you open your hot-water taps and there's a demand to heat the faucet-water .
Re the outdoor temperature sensor, mine says "0" but i have not set this up.  I'm actually wondering what the correct way is to set this up as well.  What i notice is yours is set to `MQTT`, so i'd have to ask, have you configured the right topic to be updated in order to inform the mqtt-querying system about the temperature outside? I would like to do this myself as well @diyless any guidance?

diyless, 2025-01-02:
MQTT is for indoor temperature sensor. MQTT outdoor temp is available only in test firmware (v2.2) which is currently available for a manual install only via web-flasher and a usb cable connection.

k-janssens, 2025-01-02:
Yes, that is a graph from home assistant for the DHW, the time it went up was at the moment I took a shower. And today I had a spike again at the moment I used some hot water in the kitchen.
I expect to see the water temperature from within the boiler but could be this is normal, at the moment everything seems to working OK.

My main concern is if these "wrong" numbers, especially the outdoor sensor as this can be taken into account for the CH water temperature calculation (?).

MQTT is setup correctly, all numbers are shown correctly in home assistant and the -34C is also exactly as seen directly on the MQTT topic.
I also provide our living room temperature over MQTT and that is also working correctly.

Overal I'm  quite happy with the installation, just want to have some info on these 2 numbers and if there is any negative impact on the working for the system or not.

diyless, 2025-01-02:
@k-janssens do you have an external water tank?

k-janssens, 2025-01-02:
Yes I have a water reservoir which the system will also make sure the temperature within is at the set temp of 60C.

FYI: This happened when taking some hot water, it went up to 105C which seems a bit hot?
[image]

k-janssens, 2025-01-02:
Yes it does.

In home assistent this one: sensor.living_thermostaat_dhw_temperature

It has the same value as in the web interface for the DHW.

The other attributes in Hass:
[image]

diyless, 2025-01-02:
Ah okay
Currently we support only internal boiler DHW reportings. That's why you might see them only if DHW active. For external tank with sensor we need to investigate how to pull that data via OpenTherm

k-janssens, 2025-01-02:
So in general these numbers are not something to worry about for the correct operation of the system?

k-janssens, 2025-01-02:
If the risk of bricking is low I can that for sure (the missus doesn't like it to be cold in house :))

k-janssens, 2025-01-02:
It doesn't show much only the outdoor and indoor temp.

diyless, 2025-01-02:
And if you disconnect thermostat from the boiler - is boiler able to recognize that outside sensor and work with it properly?

k-janssens, 2025-01-02:
My CH system does not give such info, I can change settings and such but not see actual temperatures.

I'm 90% sure Nest gives the sensor reading as the number was always as my outside weather station reported where as online services are always a bit off with my location. But could be wrong though, difficult to confirm with such a closed system.

diyless, 2025-01-02:
I've checked the OpenTherm specs, there is no other way reading that temperature (neither external nor water tank).
So, it is possible that the boiler reports DHW only while DHW is active.
And outside temperature might have some custom or vendor-specific command to read, which we currently don't know. The only way to figure out would be sniffing the traffic between nest and the boiler

diyless, 2025-01-02:
You can also try alternative firmwares liek ESPHome or https://github.com/Laxilef/OTGateway to see if those are able to read the outside temperature correctly. If so - that might help us to figure out how to deal with those values

k-janssens, 2025-01-02:
It is not really a necessity for me to be able to read these 2 numbers, my concern was a wrong operation of the system but that seems not to be the case.
Maybe in time I will try out another FW but for now going to stick with yours, already happy I got rid of Nest :)

aleks-mariusz, 2025-01-02:
> possible the boiler reports DHW only while DHW is active

FWIW, @k-janssens, to be more specific it might be _your specific boiler_ model being referred to here, as my boiler shows the DHW even when the DHW isn't on:

[image]

diyless, 2025-01-08:
seems we can close it, since the available info is boiler-model dependent

closed 2025-01-08 by diyless