diyless/opentherm-thermostat3 #6: Consider changing the behaviour how PI deals with CH Min and CH Max

bligeti, 2025-01-06:
Hi,

This is rather a feature request then an issue.
I have a gas boiler with radiators and I found it usefull to set CH min to 30-35⁰C since at lower water temperature the radiators are not really able to give out the heat.

As I see the behaviour of the thermostat3, it seems to minimize the output of the PI control like this:
`op = max(oplo, min(ophi, op));` 
The code is from the esp32-wifi-thermostat.ino and I think oplo is the CH min and ophi is the CH max.

For me this results in the behaviour that eventhough the heating should be off (the thermostat is in ide) the boiler remains on as the setpoint can not go below the 30⁰C. Also in general it means (if my previous assupton is correct) that setting the CHmin/max is truncating the control range of the PI.

The prevous controller I used had this:
```
      if (pid_output == 0.0f) {
        heating_target_temperature = 10.0f;
      }
      else {
        heating_target_temperature =  pid_output * (target_temperature_high - target_temperature_low) 
        +target_temperature_low;      
      }
```
So the whole control range of the PI (0-100%) was applied between the CHmin and CHmax and when the PI output was 0 the the result was a low enough value (maybe not 0 to have freeze protection).

Please consider chaning this behaviour.
Thank You

diyless, 2025-01-06:
Thanks for the input. We might add this as an optional configurable parameter (CHMin is already present, we can add a flag to either shutdown or idle the boiler when PID goes below CHmin)

DevilRange, 2026-10-20:
Because of no response for this issue, I took matters into my own hands and I posted a firmware patch and a small tool for it here: https://github.com/diyless/opentherm-thermostat3/issues/31#issuecomment-6288604870