diyless/opentherm-thermostat3 #7: Can not use Thermostat3 with my LYWSD03MMC

bligeti, 2025-01-11:
Hi,

I would like to use Thermostat 3 with my LYWSD03MMC I had from earlier (not ordered from diyless.com) since I would like to avoid the behaviour that if someone turns on the display of the thermostat the detected temperature increase by about 1 °C  and takes more than 20 min to go back (wall mount, Brightness 255, Dim brightness 150, Dim time 30, Off time 60).

I flashed it with [awesome pvvx's/ATC firmware](https://github.com/pvvx/ATC_MiThermometer), tried all advertsing type and adflag combination. Tried with [ble-repeater](https://github.com/diyless/ble-repeater) and while the repeater can detect the LYWSD03MMC and advertise it, the thermostat does not find anything, neither via screen or web interface.

What am I missing?

THank you,
Balint

diyless, 2025-01-11:
What version of the ATC firmware is used?

bligeti, 2025-01-11:
4.9

diyless, 2025-01-11:
can you try a 4.6 version? format should be PVVX with flags enabled

bligeti, 2025-01-11:
With 4.6 it is working! Thanks a lot for the great support!
(Worth to note that the [ble-repeater](https://github.com/diyless/ble-repeater) does not seems to work with Thermostat3. It is expecting a 19 byte lenght advertisement while on the 46 with flags enabled the lenght is 22 byte and also the sent advertizement       has to be tweaked as well.)

closed 2025-01-11 by bligeti

diyless, 2025-01-11:
We will investigate what's changed in 4.9, maybe its possible to add support back.
And regarding the repeater - it might be a bit outdated, since all humanity from our company (all 2 members) were working on the thermostat3 itself to be released

bligeti, 2025-01-11:
Working with ATC_v46 fw, Advertisng type: PVVX (Custom), Adflags : enabled

diyless, 2025-01-11:
BTW did you have to connect wires to flash ATC? or do you have an older revision which was capable of OTA flashing?

bligeti, 2025-01-11:
OTA, the Hw version is 1.5

DIYLESS Electronics ***@***.***> ezt írta (időpont: 2025.
jan. 11., Szo 15:34):

bligeti, 2025-01-12:
Wanted to check which ATC version introduced the incompatbility, so started to ugrade from 46  to 47 then 48 and finally to 49.
To my biggest surprise after the upgrade to 49, it is still working. Also if I start the BLE scan on the web ui, it seems it finds it.
It's seems like problem would be with the initial detecton only.

diyless, 2025-01-12:
We had a simmilar issue, if flashing 4.6, and then newer - its working, if the new from start - it's not.