diyless/opentherm-thermostat3 #15: Reducing boiler cycling

saaj, 2025-04-05:
I have this situation in the morning. I have Intergas Kombi Kompakt 28/24 boiler and the thermostat's heating algorithm is PID with default settings (*CH Max* set to 65 °C in the screenshots below – I'm experimenting with lower flow temperature – but the same pattern occurs at the default 80 °C).

It looks like that *CH Setpoint* is set correctly by the thermostat, but when it drops below 35 °C the boiler starts cycling. Can you recommend settings for the thermostat (or the boiler) to reduce cycling? Or is it a bug (e.g. *CH Setpoint* should not be set to a temperature that the boiler can't modulate, which possibly can be read from OpenTherm)?

[image]

[image]

Boiler defaults (page 40 of [the manual][1]):

<details>
<summary>defaults</summary>

[image]
</details>

[1]: https://www.intergasheating.co.uk/app/uploads/2018/04/Combi-Compact-HRE-Installation-manual-88287806.pdf

ZHNovell, 2025-04-06:
Set the max modulation parameter to about 15% and I think you will have the same situation as me. [https://github.com/diyless/opentherm-thermostat3/issues/12](url)

saaj, 2025-04-21:
I did a couple tweaks to the boiler and thermostat. Here are the results. The temperature outside during all the 3 days was comparable, (within; night-day) 5—15°C.

<details>
<summary>Before (sunny)</summary>

[image]
[image]
</details>

<details>
<summary>After (sunny)</summary>

[image]
[image]
</details>

<details>
<summary>After (cloudy)</summary>

[image]
[image]
</details>

Changes to the boiler (factory settings):

| Setting                                         | New value | Factory value  |
| ----------------------------------------------- | --------- | -------------- | 
| Minimal CH RPM *                                | 25%       | 30%            |
| Maximum CH power set **                         | 55%       | 70%            |
| Min. supply temperature during OpenTherm demand | 10°C      | 40°C           | 

*\* That must be the minimal yield*
*\*\* I didn't change it, the installer must have done it; before = after*

Changes to the thermostat:

| Setting                   | New value |
| ------------------------- | --------- | 
| CH Max                    | 60°C      |
| Relative modulation level | 40%       |

Besides the cycling becoming more coarse (on equally sunny days), on a colder (less sunny) day it's modulating most of the day (pump is active), holding the target temperature with good precision.

I think this is satisfactory (or even good result for a boiler with minimal CH power of 7.8 kW). Thus I'm closing the ticket. Feel free to reopen if you want to proactively help the user to bring some settings, readable via OpenTherm, to his attention (e.g. *Min. supply temperature during OpenTherm demand*).

closed 2025-04-21 by saaj